SpotlightKid/micropython-ftplib #1: Cannot import on esp8266

ARF1, 2018-11-10:
Thanks for making this port of ftplib available. I am having issues importing it on esp8266 though.

If I import the esp8266 port compiled to `mpy`, I get the error:
```
>>> import ftplib                                                               
Traceback (most recent call last):                                              
  File "<stdin>", line 1, in <module>                                           
ValueError: incompatible .mpy file
```
If I import the esp8266 port as a `py` file, I get the error:
```
>>> import ftplib                                                               
Traceback (most recent call last):                                              
  File "<stdin>", line 1, in <module>                                           
MemoryError: memory allocation failed, allocating %u bytes 
```
Is there any simple fix you can think of?

SpotlightKid, 2018-11-10:
If you are able to compile micropython yourself, you can try to compile the ftplib.py file to an .mpy file with a current version of mpy-cross.

ARF1, 2018-11-11:
Thanks. mpy-cross did fix the problem on esp8266.

closed 2018-11-11 by ARF1