SpotlightKid/micropython-ftplib #2: Usage on ESP32 / Error: Could not connect to...

asturkoglu, 2019-09-10:
Can I use it on ESP32 directly? I succesfully imported it on my board, but couldn't connect to FTP server.
`>> ftp = FTP(host='192.168.0.14',port=21,user='esp32',passwd='1234')`
`Traceback (most recent call last):`
` File "<stdin>", line 1, in <module>`
`  File "ftplib.py", line 206, in __init__`
`  File "ftplib.py", line 264, in connect`
`  File "ftplib.py", line 242, in _create_connection`
`Error: Could not connect to ('192.168.0.14', 21)`
I can access to the storage via Chrome, with same settings (user, passwd, etc.)

SpotlightKid, 2019-09-10:
You can enable debugging like this:

```python
from ftplib import FTP

class DebugFTP(FTP):
    debugging = 1

ftp = DebugFTP(host='192.168.0.14', port=21, user='esp32', passwd='1234')
```

This will print the exception message when the socket connection fails and may give some clue why that is.

asturkoglu, 2019-09-10:
```
>>> ftp = DebugFTP('192.168.0.14', 21, 'esp32', '1234')
requested length 2 but object has length 1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ftplib.py", line 206, in __init__
  File "ftplib.py", line 264, in connect
  File "ftplib.py", line 242, in _create_connection
Error: Could not connect to ('192.168.0.14', 21)
>>> ftp = DebugFTP('ftp://192.168.0.14', 21, 'esp32', '1234')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ftplib.py", line 206, in __init__
  File "ftplib.py", line 264, in connect
  File "ftplib.py", line 242, in _create_connection
Error: Could not connect to ('ftp://192.168.0.14', 21)
```

SpotlightKid, 2019-09-10:
The second try does not use the correct syntax for the address. Only IP addresses are allowed, no URLs.

First trial gives an important clue: 

    requested length 2 but object has length 1

It seems that the call to `_resolve_addr()` in line 226 returns something unexpected. Could you insert a `print(addrinfos) before line 227 and try again and post the output here?

asturkoglu, 2019-09-10:
Sorry, when I couldn't understand first clue, I also wanted to try it with `ftp://`
I added print function, here is the output:
```
>>> ftp = DebugFTP(host='192.168.0.14', port=21, user='esp32', passwd='1234')
[(2, 1, 0, '192.168.0.14', ('192.168.0.14', 21))]
requested length 2 but object has length 1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ftplib.py", line 206, in __init__
  File "ftplib.py", line 265, in connect
  File "ftplib.py", line 243, in _create_connection
Error: Could not connect to ('192.168.0.14', 21)
```

SpotlightKid, 2019-09-11:
Hmm, not sure why the `socket.connect)` call fails. Are you sure the network is set up correctly on your ESP32 and that address is reachable?

You might also try the `ftplib.py` version for the esp8266 in the sub-directory with the same name, which has a slightly different wrapper for the `socket` class.

SpotlightKid, 2020-04-02:
No further feedback. Assuming improper network configuration. Feel free to re-open, if you have further information.

closed 2020-04-02 by SpotlightKid