SpotlightKid/micropython-ftplib #4: ESP32 can't connect to any server

kacpo55, 2020-03-18:
Hi,

I've got the same problem as previous person... (https://github.com/SpotlightKid/micropython-ftplib/issues/2)
I can't connect to any server, debugging gives the same answer. I've tried pinging the server from my ESP32 and it works fine.

SpotlightKid, 2020-03-18:
Can you create a raw socket connection?

```py
import usocket

IP_ADDRESS = '<numerical IP address of FTP server>'

sock = usocket.socket()
sockaddr = usocket.getaddrinfo(IP_ADDRESS, 21)[0][-1]
sock.connect(sockaddr)
```

If this fails, post the error message and traceback here.

kacpo55, 2020-03-18:
I put it after connecting to wifi and recieved error:

> Traceback (most recent call last):

>  File "main.py", line 40, in <module>

> OSError: [Errno 104] ECONNRESET

SpotlightKid, 2020-03-18:
Then it seems to me the error is not with ftplib.

When you say you can ping the server from your ESP32, can you show how you do that?

Can you create other TCP connections, e.g. to an HTTP server?

kacpo55, 2020-03-18:
Ok. I've managed to connect successfully to speedtest.tele2.net.

To ping I'm using uping:

> >>> import uping
> >>> uping.ping('127.0.0.1')
> PING 127.0.0.1 (127.0.0.1): 64 data bytes
> 84 bytes from 127.0.0.1: icmp_seq=1, ttl=255, time=7.372000 ms
> 84 bytes from 127.0.0.1: icmp_seq=2, ttl=255, time=7.240000 ms
> 84 bytes from 127.0.0.1: icmp_seq=3, ttl=255, time=7.272000 ms
> 84 bytes from 127.0.0.1: icmp_seq=4, ttl=255, time=8.011000 ms
> 4 packets transmitted, 4 packets received
> (4, 4) 

> Can you create other TCP connections, e.g. to an HTTP server?
No problem to speedtest.tele2.net

kacpo55, 2020-03-18:
Know what, forgot about it. 127.0.0.1 is loopback IP... Now I'm trying to connect to my IPv4. socket gives error:

> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> OSError: [Errno 113] EHOSTUNREACH

SpotlightKid, 2020-03-19:
>> Can you create other TCP connections, e.g. to an HTTP server?
> No problem to speedtest.tele2.net

I mean on the ESP32. e.g.

```py
import usocket

# speedtest.tele2.net
IP_ADDRESS = '90.130.70.73'

sock = usocket.socket()
sockaddr = usocket.getaddrinfo(IP_ADDRESS, 80)[0][-1]
sock.connect(sockaddr)
```

SpotlightKid, 2020-04-02:
No further feedback. Assuming improper network configuration. Feel free to re-open, if you have further information.

closed 2020-04-02 by SpotlightKid