SpotlightKid/micropython-ftplib #8: ESP32 - Could not connect to ('host', port)

valenzmanu, 2020-06-04:
Hi there,

I'm getting the error mentioned in https://github.com/SpotlightKid/micropython-ftplib/issues/2 and https://github.com/SpotlightKid/micropython-ftplib/issues/4

This is my code:
```
from ftplib import FTP
ftp = FTP()
ftp.set_debuglevel(2)
ftp.connect('ftp1.at.proftpd.org')
```

I also added `print('addrinfos: ', addrinfos)` before line 227 in `ftplib.py`  as suggested in https://github.com/SpotlightKid/micropython-ftplib/issues/2#issuecomment-530065853<br>

This is what I'm getting:
```
addrinfos:  [(2, 1, 0, '', ('86.59.114.198', 21))]
requested length 2 but object has length 1

Traceback (most recent call last):
  File "<stdin>", line 44, in <module>
  File "ftplib.py", line 265, in connect
  File "ftplib.py", line 243, in _create_connection
Error: Could not connect to ('86.59.114.198', 21)
```

I was able to create a raw socket connection with this code https://github.com/SpotlightKid/micropython-ftplib/issues/4#issuecomment-600799398, it works just fine

```
import usocket

IP_ADDRESS = 'ftp1.at.proftpd.org'

sock = usocket.socket()
sockaddr = usocket.getaddrinfo(IP_ADDRESS, 21)[0][-1]
sock.connect(sockaddr)
```

This is what I get with the code above (there's no error):
```
MicroPython v0.5.0-88-gb7f0c13 on 2020-05-27; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>> 
```

Also, I've sent  messages over TCP and made HTTP requests succesfully.

SpotlightKid, 2020-06-04:
The code in the current master branch was never tested with the `esp32` port. If it works at all, you should use the version of `ftplib.py` in the `esp8266` directory.

You can also try the new version (with provisional TLS support) in the [feature/ftp-over-tls](https://github.com/SpotlightKid/micropython-ftplib/tree/feature/5/ftp-over-tls) branch.

Here's how I tested this successfully on an ESP32-WROOM dev board:

First shell session (needs [pyftpdlib](https://pypi.org/project/pyftpdlib/) Python package installed):

```
cd test
./pyftpdlib-server.py 2121
```

Second shell session:

```bash
for py in ftplib ftplibtsl ftpadvanced; do
    mpy-cross esp/${py}.py
    rshell -p /dev/ttyUSB0 cp esp/${py}.mpy /pyboard
done

cd test
pyboard --device /dev/ttyUSB0 esp/test_ftplib.py
```

valenzmanu, 2020-06-05:
Thanks for your response.

I tried using `ftplib.py` file in the `esp8266` directory but got the error below. I assume this version won't be compatible with my board.

``````
Traceback (most recent call last):
  File "<stdin>", line 44, in <module>
  File "ftplib.py", line 220, in connect
  File "ftplib.py", line 89, in __getattr__
AttributeError: 'socket' object has no attribute 'makefile'
``````

I'll try the new version.  Just a question, where should I put the wifi connection code? should I add it to the `test_ftplib.py` file or `boot.py`?

SpotlightKid, 2020-06-05:
You can just connect to the ESP board before the test and establish the connection manually, then leave the REPL without resetting the board. The connection will stay up as long as you don't reset the board.

The way I do it, I have [this module](https://github.com/SpotlightKid/picoredis/blob/master/examples/netconfig.py) (or a similar version) copied to the board and also a simple JSON configuration file for it in the format show below. Then I just use `from netconfig import connect; connect('netconfig.json')` to fire up the connection.

```json
{
    "ssid": "My WLAN",
    "password": "XXXXXXXX",
    "timeout": 8
}
```

SpotlightKid, 2020-06-09:
@valenzmanu: How did the `feature/ftp-over-tls` branch work out for you?

val15072, 2020-06-22:
@SpotlightKid  sorry for the delay. I tried the `feature/ftp-over-tls` branch, but got this error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test_ftplib.py", line 5, in <module>
  File "ftplib.py", line 160, in __init__
  File "ftplib.py", line 220, in connect
  File "ftplib.py", line 89, in __getattr__
AttributeError: 'socket' object has no attribute 'makefile'
```

valenzmanu, 2020-06-22:
@SpotlightKid sorry for the delay. I tried the feature/ftp-over-tls branch, but got this error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test_ftplib.py", line 5, in <module>
  File "ftplib.py", line 160, in __init__
  File "ftplib.py", line 220, in connect
  File "ftplib.py", line 89, in __getattr__
AttributeError: 'socket' object has no attribute 'makefile'
```

SpotlightKid, 2020-06-22:
You seem to have an older version of micropython your ESP32. In newer versions,, `usocket.socket` object should have a [makefile](http://docs.micropython.org/en/latest/library/usocket.html#usocket.socket.makefile) method:

https://github.com/micropython/micropython/blob/13ad1a4f06f2d114c821279f58fa7462e8bd0d41/ports/esp32/modsocket.c#L759

Are you able to try with a newer firmware version? Preferably a daily build from here:

https://www.micropython.org/download/esp32/

Nevertheless, I should add a workaround for a missing `makefile` method like I did in other places of the file. I think the ESP8266 port still needs that. These inconsistencies between the ports can get quite annoying sometimes.

valenzmanu, 2020-06-23:
Since I'm using [this board](https://www.seeedstudio.com/Sipeed-Maixduino-Kit-for-RISC-V-AI-IoT-p-4047.html) and not a standalone ESP32, I don't think I can use that build (but not sure). I'm using the [v0.5.0](https://github.com/sipeed/MaixPy/releases) of this [micropython port](https://github.com/sipeed/MaixPy)

SpotlightKid, 2020-06-24:
I just pushed a small fix that should allow `esp/ftplib.py` to be used with firmware versions where `socket.socket.makefile` is not supported.

Obviously I can't test this on your custom board / firmware, so please try it out and let me know, if that works for you.

valenzmanu, 2020-06-25:
Thanks!

I was able to login:

```
./pyftpdlib-server.py -p 2121
[I 2020-06-25 14:56:22] concurrency model: async
[I 2020-06-25 14:56:22] masquerade (NAT) address: None
[I 2020-06-25 14:56:22] passive ports: None
[I 2020-06-25 14:56:22] >>> starting FTP server on 0.0.0.0:2121, pid=2963 <<<
[I 2020-06-25 14:56:49] 192.168.0.138:56442-[] FTP session opened (connect)
[I 2020-06-25 14:56:53] 192.168.0.138:56442-[joedoe] USER 'joedoe' logged in.
```

However, I got an error in this line `files = ftp.nlst()` when executing `test_ftplib.py `:

```
File "ftpadvanced.py", line 18, in nlst
  File "ftplib.py", line 506, in retrlines
  File "ftplib.py", line 434, in ntransfercmd
  File "ftplib.py", line 420, in ntransfercmd
  File "ftplib.py", line 292, in sendcmd
  File "ftplib.py", line 257, in getresp
  File "ftplib.py", line 243, in getmultiline
  File "ftplib.py", line 235, in getline
EOFError: 
MicroPython v0.5.0-95-g05c4e70 on 2020-06-05; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>> 
```

SpotlightKid, 2020-06-25:
This indicates that the connection was closed while the client waited for a response to the NLST command. Though it is curious that the server output does not show anything about this. It's hard to say what caused this. I'm a unsure now how to help you further without access to the same firmware or board as you're using.

valenzmanu, 2020-06-26:
I see, I'll try to debug it. Ultimately what I need is just to upload a `.jpg` file, I don't really need the ability to list all the files in the FTP server. I would like to try to upload a file, if it works, would be more than enough for my application.

I saw the `test_upload.py` and `ftpupload.py` files outside the `esp` directory. How can I upload a file using a simple FTP connection? (Not over TLS), should I use the `upload` function?

SpotlightKid, 2020-06-26:
> should I use the upload function?

Yes. Just make sure that you install / import the right `ftplib.py` version.

SpotlightKid, 2020-07-21:
Closing this due to no further feedback. Please feel free to re-open the issue if you have more information.

closed 2020-07-21 by SpotlightKid