SpotlightKid/micropython-ftplib #10: error when get file sizes

m986883511, 2021-03-15:
thank for your project, I can download files , but I can not get file sizes, can your help me.

Traceback (most recent call last):
  File "main.py", line 37, in <module>
  File "ftplib.py", line 635, in size
  File "ftplib.py", line 292, in sendcmd
  File "ftplib.py", line 262, in getresp
Error: 550 SIZE not allowed in ASCII mode.

code:

from ftpadvanced import AdvancedFTP

PORT = 21
ftp = AdvancedFTP('192.168.2.232', PORT)
ftp.login()
files = ftp.nlst()
print(files)
print('sizes=', ftp.size('aaaaaa.dxf'))#error

SpotlightKid, 2021-03-15:
Try using `ftp.voidcmd('TYPE I')` before the `ftp.size(...)` call.

You can also try using the `ftp.mlsd()` method, if the server supports it.

m986883511, 2021-03-15:
thank you , it works.

closed 2021-03-15 by m986883511