SpotlightKid/micropython-ftplib #11: ftp download success, but I dont known where is the file

m986883511, 2021-03-15:
I test the download and upload function, upload func works,but the download func have same problem, can not find the download file.

code:
`result = ftp.retrbinary('RETR aaaaaa.dxf', progress)
print('ftp download aaaaaa.dxf return code=', result)

with open('README.pdf', 'rb') as fp:
    result = ftp.storbinary('STOR README.pdf', fp)
    print('ftp upload README.pdf return code=', result)`

serial print:

ftp download aaaaaa.dxf return code= 226 Transfer complete.
ftp upload README.pdf return code= 226 Transfer complete.
MicroPython v1.13 on 2020-09-02; ESP32 module with ESP32
Type "help()" for more information.

>>> import os
» os.listdir()

« os.listdir()
['README.pdf', 'esp', 'ftp.py', 'ftp_pycom.py', 'ftp_thread.py', 'ftpadvanced.py', 'ftplib.py', 'ftplibtls.py', 'main.py', 'uftpd.py']

where is the download file aaaaaa.dxf?

SpotlightKid, 2021-03-15:
Nowhere, you need to save the received data yourself into file in the callback function (the one you called `progress`).

m986883511, 2021-03-15:
`
fileToWrite = open("aaaaaa.dxf", "wb")
def handleDownload(block):
    fileToWrite.write(block)

result = ftp.retrbinary('RETR aaaaaa.dxf', handleDownload)
print('ftp download aaaaaa.dxf return code=', result)
`

thank you， it works！

closed 2021-03-15 by m986883511