SpotlightKid/micropython-ftplib #12: Trouble installing this on rpi pico 2 w

rickavila, 2022-08-28:
When I run setup.py on a new pico 2 W it gives an "ImportError: no module names 'setuptools'". I get stuck here. I am unable to find setuptools for micropython. Any assistance is appreciated.

SpotlightKid, 2022-08-28:
`setup.py` is only for CPython/PyPI or the `unix` port . For micropython on microntroller boards, just copy `ftplib.py` and optionally `ftplibtls.py` and `ftpupload.py` to the flash file system of your board. You can also [compile them](http://docs.micropython.org/en/latest/reference/mpyfiles.html) with `mpy-cross` first and copy the resulting `*.mpy` files instead.

nivaneram, 2022-09-06:
Traceback (most recent call last):
  File "<stdin>", line 13, in <module>
  File "ftplibtls.py", line 59, in __init__
  File "ftplib.py", line 216, in __init__
  File "ftplib.py", line 274, in connect
  File "ftplib.py", line 252, in _create_connection
Error: Could not connect to ('***.**.**.***', 21)

Note :-(ip is hidden with *)  Working on Raspberry pi pico w

SpotlightKid, 2022-09-06:
Please use `.set_debuglevel(2)` on your `FTP` instance before connecting - i.e. don't pass a host/IP on instantiation use `.connect(host)`. Then report the output here.

The second error is likely a result of the first one, because there is no connection, the instance does not have an `af` attribute yet.

SpotlightKid, 2022-09-06:
BTW, I don't have a Rasperry Pico W, so I haven't tested this library on the pico port and I am not able to at the moment.

nivaneram, 2022-09-24:
@SpotlightKid Thanks for the reply it is working properly

closed 2022-09-24 by SpotlightKid