SpotlightKid/micropython-ftplib #13: issues on RPi Pico W

polebridge, 2022-12-04:
I ran into 2 issues which i "fixed?"  I'm new to Python so these fixes are not authoritative.
In ftplib in "def connect(self, addr):"
the code reads:  self._sock.connect(_resolve_addr(addr))
but  this gives an error deep in \micropython-1.19.1\py\obj.c  "requested length 2 but object has length 1" 
_resolve_addr(addr) returns an array with 1 element, but connect wants 2 - IP and port
the addr is already resolved.  
I "fixed" by changing to:  self._sock.connect(addr)

Then i encountered a python error in getmultiline at "line = line + (b'\n' + nextline)"
but line and nextline are both str, Python moans about concat a byte with a string
I "fixed" by:  line = line + "\n" + nextline)

With those two changes i was able to log onto a server, list files, change dir, upload a file, quit, and close.

What is your opinion of my fixes?

SpotlightKid, 2022-12-05:
The `socket` module behaves differently on the various MicroPython ports. On most, `connect` expects a the return value from `socket.getaddrinfo`:

http://docs.micropython.org/en/latest/library/socket.html#socket.socket.connect

Apparently this is not the case on the Raspberry Pico W port.

Still, any changes to this module would need to work with all ports supported so far.

Since I don't have a Pico W, I can't test this. Probably there are other subtle differences, like returning `str` instead of `bytes` from some methods, as you have observed.

polebridge, 2022-12-05:
Thanks.  I understand the need for compatibility and portability.
And thanks for the link to the docs on micropython.org

But i see in those docs an example 
s.connect(socket.getaddrinfo('www.micropython.org', 80)[0][-1])
so it's not connecting to the whole result of getaddrinfo, but only to the last element, a tuple of IP and port.

I notice that Python 3.10 on Windows and on debian will not accept the list element from getaddrinfo, but diagnose 
TypeError: connect(): AF_INET address must be tuple, not list".  So perhaps the MicroPython port to Pi is closer to standard?

Also, the "def getline(self):"  function  calls file.readline to get bytes, but then calls the "decode()" method, which turns the bytes into a str, which getline returns to "getmultiline".  getmultiline is attempting to concatenate strings and a byte of b'\n', and gets the diagnosis "TypeError: can't concat str to bytes"

-Rick Murphy

JMFUZ, 2023-02-28:
Hi
@polebridge : Did you find a solution?
Thx

SpotlightKid, 2023-02-28:
> so it's not connecting to the whole result of getaddrinfo, but only to the last element, a tuple of IP and port.

[This line](https://github.com/SpotlightKid/micropython-ftplib/blob/3bca4d23100da05dbe933c39d07bd1439da57806/ftplib.py#L237) unpacks the result.