SpotlightKid/micropython-ftplib #15: I can't connect to the ftp server (ESP32-WROOM-32).

DavideLentini, 2023-10-05:
Hi , I don't know if I am doing something wrong .
I have an esp-32 :
https://www.az-delivery.de/it/products/esp32-developmentboard
With micropython installed , and I am using Thonny .
I loaded on the esp32 , the` ftplib.py `file .
And then in the` boot.py` file I added the line : 

`from ftplib import FTP`

In main.py , after I connect to the wifi network , I try to access an ftp server local on the ip 192.168.1.1 

the part of code dedicated to ftp in main.py is :

```
ftp= FTP('192.168.1.1',21,'user','password')
print('done')

```
When I try to run the code an error message comes up : 

`can't connect to('192.168.1.1',21)
`
I confirm that the server is working , and that the esp32 is able to ping the ip 192.168.1.1

SpotlightKid, 2023-10-05:
Hard to diagnose from here. Try:

```py
from ftplib import FTP

class DebuggingFTP(FTP):
    debugging = True

ftp = DebuggingFTP('192.168.1.1',21,'user','password')
```

and that should print the actual exceptions raised when connecting fails.

Edit https://github.com/SpotlightKid/micropython-ftplib/blob/master/ftplib.py#L242 if you want a traceback.

SpotlightKid, 2023-10-05:
> i have to use the library inside the esp folder right?

Correct.

DavideLentini, 2023-10-05:
Thanks ! I'll try tonight again , with the library in the esp folder :)

DavideLentini, 2023-10-07:
It worked , it was my error . Sorry for the issue

closed 2023-10-07 by DavideLentini