SpotlightKid/micropython-ftplib #18: FTP_TLS error

JMFUS, 2023-12-13:
Hello
Here is my code :
import network
from ftplibtls import FTP_TLS

# Connexion au reseau
status_reseau = connect_wifi_local("NETWORK", "123456789")[1]

if status_reseau == 3:	# si on est connecté à internet via le reseau local

    # Connexion au serveur FTPS
    try:
        
        ftp_tls = FTP_TLS()
        print("1")
        ftp_tls.connect(adresse_serveur, port)
        print("2")
        ftp_tls.login(utilisateur_1, mot_de_passe_1)
        print("3")
        ftp_tls.prot_p()
        print("4")
        files = ftp_tls.nlst()
        print(files)
        
    except Exception as e:
        erreur = str(e)
        print("Erreur FTPS : " + str(e))

But I have the following error on the login line :

Erreur FTPS : extra keyword arguments given

Someone can help me?

Thx

SpotlightKid, 2023-12-13:
Please remove the wrapping of the code in a `try/except` block and run it again. That way you will get a full exception traceback, which shows where the error happens exactly.

JMFUS, 2023-12-13:
Thx for your answer
Now I have :
Traceback (most recent call last):
  File "<stdin>", line 142, in <module>
  File "ftplibtls.py", line 63, in login
  File "ftplibtls.py", line 80, in auth
TypeError: extra keyword arguments given

SpotlightKid, 2023-12-13:
Which Micropython port are you using? Unix, stm32, esp2866, esp32, etc. ?

SpotlightKid, 2023-12-13:
The Pico W port is unfortunately *not supported* yet, since I don't have a Pico W board to test against.

The API of the `socket` and `ssl` modules is slightly different on each MicroPython port and it seems that the [`ssl.wrap_socket`](https://docs.micropython.org/en/latest/library/ssl.html#ssl.ssl.wrap_socket) function on the Pico W port does not support all the keyword arguments used by the current `ftplibtls` code.

You can try whether the variant for the ESP boards in the `esp/ftplib.py` and `esp/ftplibtls.py` files works on the Pico W as well, but they may very well do not.

Alternatively, you can try changing lines 77-80 of `ftplibtls.py` to:

```py
        wrapped = ssl.wrap_socket(sock,
                                  keyfile=self.keyfile,
                                  certfile=self.certfile)
```

i.e. remove the `cert_reqs` keyword argument.

---

P.S.: If someone would sponsor me by [sending me a Pico W board](https://www.berrybase.de/raspberry-pi-pico-w-rp2040-wlan-mikrocontroller-board), I could "officially" support the Pico W port :grinning:

JMFUS, 2023-12-13:
I changed the lines 77-80 of the ftplibtls.py but the result is the same :(

SpotlightKid, 2023-12-13:
Try removing the `keyfile` and `certfile` arguments as well.

Note: of course this means that no certificate validation will take place, which partly defeats the purpose of using TLS. So use at your own risk!

JMFUS, 2023-12-13:
I can connect to my FTPS server but I can't list the files of my directory
line =  b'220-FileZilla Server 0.9.60 beta\r\n'
line =  b'220-written by Tim Kosse (tim.kosse@filezilla-project.org)\r\n'
line =  b'220 Please visit https://filezilla-project.org/\r\n'
line =  b'234 Using authentication type TLS\r\n'
line =  b'331 Password required for jeanmarcf\r\n'
line =  b'230 Logged on\r\n'
line =  b'200 PBSZ=0\r\n'
line =  b'200 Protection level set to P\r\n'
line =  b'200 Type set to A\r\n'
line =  b'227 Entering Passive Mode (88,122,12,99,128,235)\r\n'
line =  b'150 Opening data channel for directory listing of "/"\r\n'
resp : 150 Opening data channel for directory listing of "/"
sock =  <socket state=3 timeout=-1 incoming=0 off=0>
Traceback (most recent call last):
  File "<stdin>", line 122, in <module>
  File "ftplib.py", line 509, in retrlines
  File "ftplibtls.py", line 136, in ntransfercmd
OSError: stream operation not supported

SpotlightKid, 2023-12-13:
This seems to be another incompatibility of the `socket' or `ssl` module on the RP2.

I can only repeat: the RP2 is not supported and unless I can get my hands on a Pico W board, I can't do nothing about it.

I suggest looking for help on the MicroPython forum until then:

https://forum.micropython.org/viewforum.php?f=21

SpotlightKid, 2024-02-08:
I finally got my hands on a Raspberry Pi Pico W board and I believe this issue is now fixed and FTP over TLS also works on the `rp2` port.

@JMFUS I would appreciate if you could test this and confirm that it works for you.
 
Please see the README for usage notes.

SpotlightKid, 2024-02-26:
Assuming this to be fixed by 6c19a8446c0c0423267ecb79a949e14f05efb613.

Feel free to re-open, if the error persists with the latest version of the code in this repo.

If you have any other problems using this on the PR2 (or any other) port, please open a new ticket.

closed 2024-02-26 by SpotlightKid