SpotlightKid/micropython-ftplib #19: OSError: [Errno 12] ENOMEM

JuanEnHD, 2024-02-22:
Hello!
I'm trying to execute the example code:

```py
>>> from ftplibtls import FTP_TLS
>>> ftp = FTP_TLS('servername')  # default port 21
>>> ftp.login('username', 'password')
>>> ftp.prot_p()
>>> ftp.retrlines('LIST')
```

But I'm getting this error:

Traceback (most recent call last):
  File "boot.py", line 71, in <module>
  File "modules/ftplib.py", line 510, in retrlines
  File "modules/ftplibtls.py", line 110, in ntransfercmd
  File "modules/ftplibtls.py", line 126, in _wrap_socket
OSError: [Errno 12] ENOMEM
MicroPython v1.22.1 on 2024-01-05; Generic ESP32 module with ESP32

The connection is successful, I don't have problems with the other functions, only with `dir()`, in addition to this one.

SpotlightKid, 2024-02-22:
Which board do you have exactly?

Can you try the version in the `esp` directory?

Also, try this example, which retrieves a plain file list and should use less memory:

```py
ftp = FTP_TLS('localhost')
ftp.login('username', 'password')
ftp.prot_p()
files = []
ftp.retrlines('LIST', callback=files.append)
print("\n".join(files))
```

JuanEnHD, 2024-02-23:
I'm using the ESP32 DEVKIT V1 board, and I'm also trying it with the version that is inside the `esp` folder. I have tested with the code provided, but the error still persists.

SpotlightKid, 2024-02-23:
How many files are in the remote directory you are trying to list?

SpotlightKid, 2024-02-23:
Also, please make sure you use the latest version of the code from the repository. The line numbers from the exception traceback you posted indicates that you are using an older version of the code.

Also, it helps to *only* copy compile `.mpy` files to the board and import those. The `install.sh` does that for you. Make sure you have `rshell` and `mpy-cross` installed on the computer where you run the installation and then run:

```con
PORT=/dev/tty/USB0 BAUD=115200 DESTDIR=/pyboard/lib ./install.sh
```

JuanEnHD, 2024-02-23:
"Hello!

I've already installed rshell and mpy-cross, but I really have no idea where to run "PORT=/dev/tty/USB0 BAUD=115200 DESTDIR=/pyboard/lib ./install.sh". Could you be more specific? Sorry, I'm an amateur."

SpotlightKid, 2024-02-23:
> I really have no idea where to run "PORT=/dev/tty/USB0 BAUD=115200 DESTDIR=/pyboard/lib ./install.sh"

In your shell in a terminal window inside the root directory of this repository.

JuanEnHD, 2024-02-26:
Hello again!
Thank you very much for the help :D. Copy only compiled `.mpy` files to the board has solved my problem. Truly, thank you.

JuanEnHD, 2024-02-26:
Hello again!
Thank you very much for the help :D. Copying only compiled .mpy files to the board has solved my problem. Truly, thank you

closed 2024-02-26 by JuanEnHD