SpotlightKid/micropython-ftplib #24: Pico W port: sponsorship

scruss, 2024-03-15:
> P.S.: If someone would sponsor me by [sending me a Pico W board](https://www.berrybase.de/raspberry-pi-pico-w-rp2040-wlan-mikrocontroller-board), I could "officially" support the Pico W port :grinning:

_Originally posted by @SpotlightKid in https://github.com/SpotlightKid/micropython-ftplib/issues/18#issuecomment-1853916612_

It would be slightly impractical for me to ship one to you from Canada, so what level of one-off sponsorship would you need to buy one from the [BerryBase link](https://www.berrybase.de/raspberry-pi-pico-w-rp2040-wlan-mikrocontroller-board) you referenced?

SpotlightKid, 2024-03-15:
Hey, @scruss Thanks for the offer, but I already got my hands on a Raspberry Pi Pico WT in the meantime and already implemented support for it in this library with commit 6c19a84. Check out the latest version from the repo and let me know if there any problems.

I do have a GitHub sponsoring page: https://github.com/sponsors/SpotlightKid

SpotlightKid, 2024-03-15:
P.S. I would love to test my MicroPython networking libraries (https://github.com/SpotlightKid/mrequests, https://github.com/SpotlightKid/micropython-ftplib, https://github.com/SpotlightKid/micropython-osc) on the [Seeed XIAO ESP32C3](https://www.berrybase.de/seeed-xiao-esp32c3-winziges-mcu-board-mit-wlan-und-ble).

That would cost me about 13 € (incl. shipping) to order at BerryBase.

scruss, 2024-03-15:
Okay, consider it done!

closed 2024-03-15 by scruss

SpotlightKid, 2024-03-15:
Awesome, thanks @scruss !!