SpotlightKid/micropython-ftplib #25: I found ftplib is incompatible to Micropython's "requests" when used on the ESP32

PV-autarky-optimizer, 2025-05-02:
I tested micropython-ftplib with Micropython on my ESP32 and it works well. That is the good News.
 
I found `ftplib` is incompatible to Micropython's intrinsic library `requests` for ESP32. It's a pitty.

My Micropython code is like this:
```python
import network                  # This library provides WLAN
from ftplib import FTP as FTP   # FTP Client from github.com/SpotlightKid
import requests as requests     # This library belongs to Micropython and carries out GET requests in the HTTP protocol. 
... setup_WLAN() ... 
response = requests.get('https://s18.myenergi.net/cgi-jstatus-*', headers = {'User-Agent': '...'}, auth = None, timeout=20)
``` 

As soon as `ftplib` is imported, `requests.get` fails to access the intrnet and after timeout it returns the error `[Errno 116] ETIMEDOUT`.

This incompatibility occures both with ftplib.py and esp/ftplib.py

By the way, in ftplib.py, line 353, fix the bug by changing `b'\n'` to `'\n'`.

SpotlightKid, 2025-05-03:
> By the way, in ftplib.py, line 353, fix the bug by changing `b'\n'` to `'\n'`.

Fixed in 82f32b45f613057669c647c0a385b60d17de7914.

SpotlightKid, 2025-05-03:
> As soon as `ftplib` is imported, `requests.get` fails to access the internet and after timeout it returns the error `[Errno 116] ETIMEDOUT`.

I can't reproduce that. Maybe this is a memory issue? Have tried compiling the ftplib modules with `mpy` and installing those instead of the `.py` files (see the  provided install scripts)?

PV-autarky-optimizer, 2025-05-06:
Dear SpitlightKid,

your hint is right. First, I learned to install and operate the compiler `mpy-cross`. Then I converted `ftplib.py` to `ftplib.mpy`. Now my problem changed. Now the error message doesn't stop the 1st HTTP-request of my script. Now 3 times of HTTP-requests succeed and the error appears at the 4th time.

I was wrong, when I said `ftplib` to be incompatible to Micropython's `request`. Instead, the RAM in my ESP32 is used up. Maybe you can recommend me some quick tutorials in English or German on monitoring the different parts of RAM and on garbage collection. The next weeks or months, I'll go on compiling libraries and I'll compare the RAM consumption of Micropython's `requests` library with `SpotlightKid/myrequests`. The latter economically uses one `socket` class both for FTP and for HTTP-requests.

With the ESP32, I actually just wanted to quickly compare the electricity consumption of my private home with the so-called green electricity index "Grünstromindex". This is a public computer calculation that shows live, when in Germany the power generation is from renewables and where the power lines are jammed. This goal is now a long way into the future.

massimosala, 2025-05-24:
Hi

Do you solve the FTP problems?

If am developing a set of micro services for micros:
https://saltlake71.eu/

There is already the possibility to send data from the micro to this server, which converts the data and sends it via email:
https://saltlake71.eu/email.html

Eventually I can add the fetature to send the data by FTP.

Ciao
Massimo